the78mole/esphome_components #1: WROOM-32D pin association

zokl, 2022-08-15:
Hello @the78mole.

I checked your PCD design and have a question about the serial and I2C pin assignment. You use U1TX and U1RX on GPIO pins 2 and 4. In default, wiring uses UART1 GPIO 9 and 10. The same for I2C. You use GPIO 13 and 16, but the original is GPIO 21 and 22. What is the reason that you select these pins?

P.S. Thank you for very good job!

the78mole, 2022-08-15:
Hi zokl,

my latest schematic looks like this:
[image]

This uses Pin 24 (IO2) for TX to the Buderus and Pin 26 (IO4) for receiving from the Buderus.
The pins I selected for the UART in my version 0.0.1 have not been working. This is fixed in 0.0.4 and 0.0.5.

For the I2C I had no eye on the hardware support at that time and I did not test it. Probably, it will be emulated in software by arduino/ESPhome. But you are totally right with IO21 and IO22. Maybe I'll correct it in 0.0.6.

For pin assigment, I followed the Pin-Selection-Helper-Excel-Sheet from Andreas Spiess (https://www.youtube.com/c/AndreasSpiess).

I plan to release the 0.0.5 (or 0.0.6) and it will also be available as a fully assembled board that can be ordered on tindie sooner or later.
Unfortunately, I can not find any time to work on it :-(

Regards,
Daniel

zokl, 2022-08-15:
> Hi zokl,
> 
> my latest schematic looks like this: [image]
> 
> This uses Pin 24 (IO2) for TX to the Buderus and Pin 26 (IO4) for receiving from the Buderus. The pins I selected for the UART in my version 0.0.1 have not been working. This is fixed in 0.0.4 and 0.0.5.
> 
> For the I2C I had no eye on the hardware support at that time and I did not test it. Probably, it will be emulated in software by arduino/ESPhome. But you are totally right with IO21 and IO22. Maybe I'll correct it in 0.0.6.
> 
> For pin assigment (for the corrected UARTs), I followed the Pin-Selection-Helper-Excel-Sheet from Andreas Spiess (https://www.youtube.com/c/AndreasSpiess).
> 
> I plan to release the 0.0.5 (or 0.0.6) and it will also be available as a fully assembled board that can be ordered on tindie sooner or later. Unfortunately, I can not find any time to work on it :-(
> 
> Regards, Daniel

I use followed schematics, and the description on your schematic does not correspond with it.  I will check the pin possition in detail. 

[image]

closed 2022-11-28 by the78mole