the78mole/esphome_components #2: 1-Wire Bus support

zokl, 2022-08-15:
Hi @the78mole.  I would like to use your board but i need to make small modiffication. I would like to add there 1-wire iface for sensors to the boiler and end it by terminals: https://de.farnell.com/phoenix-contact/pt-1-5-8-pvh-5-0/terminal-block-pluggable-8pos/dp/1793552?gclid=Cj0KCQjw3eeXBhD7ARIsAHjssr_5iBLzo18RP7qZRayOeb2QF7z_NSONAUBIKK1RHS0NTV0xu_wFagAaAnYJEALw_wcB&mckv=_dc%7Cpcrid%7C%7Cplid%7C%7Ckword%7C%7Cmatch%7C%7Cslid%7C%7Cproduct%7C1793552%7Cpgrid%7C%7Cptaid%7C%7C&CMP=KNC-GCZ-GEN-SHOPPING-PMax&gross_price=true

In your last desing 0.0.4 you fix the issue with power and module works without restarts?

the78mole, 2022-08-15:
Hi zokl,

in my Installation, it worked always without restarts, but a few people reportet problems. With the larger capacitors, the problem seems to be vanished completely, but I can not guarantee it.

You could use the P3, P4 and P7 to P12 signals. They are easily accessible on the jumper row and the only modifications you need to do are about the components used for filtering and damping the analog signals:
[image]

Should be an easy task to add the interfacing for onewire (pull-up + optional low-pass).

Regards,
Daniel

the78mole, 2022-08-15:
Here you could find some information about the best interfacing practices for 1-wire:
https://www.maximintegrated.com/en/design/technical-documents/tutorials/1/148.html

Especially the "Appendix A. Improved CPU Bus Interface" and Appendix B:
[image]

zokl, 2022-08-15:
I am considering redoing the board and using terminal strips instead of PLS strips. They will be easier to handle the wires. Just making sure I wouldn't do any unnecessary work if you already have some of this planned.

the78mole, 2022-08-15:
Du you need the latest KiCad-Project?

zokl, 2022-08-15:
> Du you need the latest KiCad-Project?

If you push it to github, I made or a pull request.

the78mole, 2022-08-15:
The HW ist hosted on Gitlab...
https://gitlab.com/the78mole/buderus-km271-mm

zokl, 2022-08-15:
> The HW ist hosted on Gitlab...
> https://gitlab.com/the78mole/buderus-km271-mm

Yes I know, i forked latest version 0.0.4.

the78mole, 2022-08-15:
I just merges the MM-One (the one above, a modified version from a friend) and my repo to have the same HW state...
https://gitlab.com/the78mole/logamatic_2107_wifi_comm

closed 2022-11-28 by the78mole