the78mole/esphome_components #9: Process a single bit of a telegram for a sensor.

jensgraef, 2022-12-04:
Be able to process a single bit of a byte of a specific telegram and use that for a sensor -> read "Betriebswerte" of HK1, WW and so on.

Bascht74, 2022-12-08:
@jensgraef We need that e.g. for this: 

`        { LPWW      , true, false, SensorType::INT, 0, "Ladepumpe", "", NULL },                          // ["aus", "Ladepumpe", "Warmwasserpumpe", "beide"]`

There we need to create two binary sensors out of it: 
```
BIN 00 DEC 0 = Beide aus
BIN 01 DEC 1 = Ladepumpe an
BIN 10 DEC 2 = Zirkulationspumpe an
DIN 11 DEC 3 = Ladepumpe und Zirkulationspumpe an
```
actually this telegram is mapped to a binary sensor so only the first bit is processed (Ladepumpe an/aus).

Do you see any way to do that with the existing code?

This is the water heater of the Buderus: 
[image]

As you can see there must be something wrong with it because it is losing temperature quiet fast... 15°C in 2,6 hours !?!?
That is why I want so see the circulation that is behind the other "bit" --> Zirkulationspumpe :-) ...

jensgraef, 2022-12-08:
With the current code you could map it to a numeric sensor and do the remaining work in home assistant, e.g. with an automation that toggles two switches depending on the value of the number.

I've got a rough idea how to do it in the km217 module, but it's not yet full thought true:
1. create a new sensor type BitArray
2. create a new class for  handling those bit arrays. This class should carry a list map of bit indices to binary sensors
3. extend the python part to allow telling the module "this is a binary sensor, and it should watch bit x of this parameter
4. extend the main class to add such a binary sensor to the class defined in 2

I wont have much time until next week, so it might take some time to build this.

the78mole, 2022-12-10:
@jensgraef you name it, it is really crap to implement inside an ESPhome component.

Bascht74, 2022-12-11:
Yes, this is how modbus controller has done it. 

Maybe this could help (I don't understand it completely but it seems to follow a lot of @jensgraef points... )

https://esphome.io/components/modbus_controller.html
--> Section bitmask in the documentation

Code:
https://github.com/esphome/esphome/blob/dev/esphome/components/modbus_controller/modbus_controller.h
--> template<typename N> N mask_and_shift_by_rightbit(N data, uint32_t mask)
plus: 
https://github.com/esphome/esphome/tree/dev/esphome/components/modbus_controller/binary_sensor

Maybe we could do the same with the address and byte that has so be read?

So maybe something like this could be very flexibly, because everybody can "select" anything via his yaml code without changing the source / adding something to the source: 

```
binary_sensor:
- platform: km271_wifi
  id: some_sensor_id
  name: "Zirkulationspumpe"
  address: 0x8429
  byte: 1
  bitmask: 0x2 #bit 1
- platform: km271_wifi
  id: some_sensor_id
  name: "Ladepumpe"
  address: 0x8429
  byte: 1 # first byte after the address
  bitmask: 0x1 #bit 0
- platform: km271_wifi
  id: some_sensor_id
  name: "Absenkung_Solar"
  address: 0x8429
  byte: 1 # first byte after the address
  bitmask: 0x2 #bit 2
```

same "style" for sensors: 
```
- platform: km271_wifi
  id: raumsolltemperatur_hk1
  name: "Raumsolltemperatur HK1"
  address: 0x8004
  byte: 1 # first byte after the address
  unit_of_measurement: "°c"
  device_class: "temperature"
  state_class: "measurement"
  value_type: U_INT_DIV2
  accuracy_decimals: 1
```
This way we could even read the config variables to sensors: 
```
- platform: km271_wifi
  id: tagtemperatur_hk1
  name: "Tagtemperatur_HK1"
  address: 0x0000
  byte: 3 # third byte after the address
  unit_of_measurement: "°c"
  device_class: "temperature"
  state_class: "measurement"
  value_type: U_INT # unsigned integer
  accuracy_decimals: 1
```

It would be more generic and will need longer yaml code but you would be very flexible with the yaml without changing anything to the c++ code...

jensgraef, 2022-12-11:
#25 adds the foundation for single bit sensors and adds the circulation pump / loading pump sensors.
@Bascht74  If you like you can check out the branch ```support-single-bit-sensors``` and check if it works as expected.

If someone wants to create a pull request for splitting all the other sensors (e.g. boiler error), feel free to do so.

qschneider, 2022-12-13:
I extended km271.h and km271_params.h with more single-bit entities, if its ok i move on with binaray_sensor.py and buderus-km271.yaml

qschneider, 2022-12-13:
km271.h#

    #pragma once
  
    #include <string>
    #include <vector>
    #include <chrono>
    #include <iostream>
    #include <sys/time.h>
    #include <ctime>
    #include "3964r.h"
    #include "esphome/core/component.h"
    #include "esphome/components/sensor/sensor.h"
    #include "esphome/components/uart/uart.h"
    #include "km271_params.h"
  
  
    #define GENERATE_SENSOR_SETTER(key, parameterId, sensorTypeParam) void set_##key##_sensor(esphome::sensor::Sensor *sensor) \
      { set_sensor(parameterId, sensorTypeParam, sensor); }
    #define GENERATE_BINARY_SENSOR_SETTER(key, parameterId, sensorTypeParam) void set_##key##_binary_sensor(esphome::binary_sensor::BinarySensor *sensor) \
      { set_binary_sensor(parameterId, sensorTypeParam, sensor); }
    #define GENERATE_SWITCH_SETTER(key, parameterId, sensorTypeParam) void set_##key##_switch(BuderusParamSwitch *switch_) \
      { set_switch(parameterId, sensorTypeParam, switch_); }
    #define GENERATE_NUMBER_SETTER(key, parameterId, sensorTypeParam) void set_##key##_number(BuderusParamNumber *number) \
      { set_number(parameterId, sensorTypeParam, number); }
  
  
    namespace esphome {
    namespace KM271 {
  
    class KM271Component : public Component, public uart::UARTDevice {
    public:
      KM271Component();
  
      void loop() override;
      void dump_config() override;
  
      // Betriebswerte 1 HK1
      GENERATE_BINARY_SENSOR_SETTER(heating_circuit_1_switch_off_optimization, BW1HK1, 0);
      GENERATE_BINARY_SENSOR_SETTER(heating_circuit_1_switch_on_optimization, BW1HK1, 1);
      GENERATE_BINARY_SENSOR_SETTER(heating_circuit_1_automatic, BW1HK1, 2);
      GENERATE_BINARY_SENSOR_SETTER(heating_circuit_1_ww_priority_processing, BW1HK1, 3);
      GENERATE_BINARY_SENSOR_SETTER(heating_circuit_1_screed_drying, BW1HK1, 4);
      GENERATE_BINARY_SENSOR_SETTER(heating_circuit_1_holiday, BW1HK1, 5);
      GENERATE_BINARY_SENSOR_SETTER(heating_circuit_1_antifreeze, BW1HK1, 6);
      GENERATE_BINARY_SENSOR_SETTER(heating_circuit_1_manually, BW1HK1, 7);
      // Betriebswerte 2 HK1
      GENERATE_BINARY_SENSOR_SETTER(heating_circuit_1_summer, BW2HK1, 0);
      GENERATE_BINARY_SENSOR_SETTER(heating_circuit_1_day, BW2HK1, 1);
      GENERATE_BINARY_SENSOR_SETTER(heating_circuit_1_no_comm_with_rc, BW2HK1, 2);
      GENERATE_BINARY_SENSOR_SETTER(heating_circuit_1_rc_faulty, BW2HK1, 3);
      GENERATE_BINARY_SENSOR_SETTER(heating_circuit_1_flow_sensor_error, BW2HK1, 4);
      GENERATE_BINARY_SENSOR_SETTER(heating_circuit_1_max_flow, BW2HK1, 5);
      GENERATE_BINARY_SENSOR_SETTER(heating_circuit_1_external_fault_input, BW2HK1, 6);
      // GENERATE_BINARY_SENSOR_SETTER(heating_circuit_1_party_pause, BW2HK1, 7);
  
      GENERATE_SENSOR_SETTER(heating_circuit_1_flow_target_temperature, VSTHK1, 0);
      GENERATE_SENSOR_SETTER(heating_circuit_1_flow_temperature, VITHK1, 0);
      GENERATE_SENSOR_SETTER(heating_circuit_1_room_target_temperature, RSTHK1, 0);
      GENERATE_SENSOR_SETTER(heating_circuit_1_room_temperature, RITHK1, 0);
      GENERATE_SENSOR_SETTER(heating_circuit_1_pump_power, PLHK1, 0);
      GENERATE_SENSOR_SETTER(heating_circuit_1_mixer_position, MSHK1, 0);
      GENERATE_SENSOR_SETTER(heating_circuit_1_curve_p10, KLHK1_P10, 0);
      GENERATE_SENSOR_SETTER(heating_circuit_1_curve_0, KLHK1_P00, 0);
      GENERATE_SENSOR_SETTER(heating_circuit_1_curve_n10, KLHK1_N10, 0);
  
      // Betriebswerte 1 HK2
      GENERATE_BINARY_SENSOR_SETTER(heating_circuit_2_switch_off_optimization, BW1HK2, 0);
      GENERATE_BINARY_SENSOR_SETTER(heating_circuit_2_switch_on_optimization, BW1HK2, 1);
      GENERATE_BINARY_SENSOR_SETTER(heating_circuit_2_automatic, BW1HK2, 2);
[... 90 lines cut ...]
  
      void set_sensor(Buderus_R2017_ParameterId parameterId, uint16_t sensorTypeParam, esphome::sensor::Sensor *sensor);
      void set_binary_sensor(Buderus_R2017_ParameterId parameterId, uint16_t sensorTypeParam, esphome::binary_sensor::BinarySensor *sensor);
      void set_switch(Buderus_R2017_ParameterId parameterId, uint16_t sensorTypeParam, BuderusParamSwitch *switch_);
      void set_number(Buderus_R2017_ParameterId parameterId, uint16_t sensorTypeParam, BuderusParamNumber *number);
  
      void process_incoming_byte(uint8_t c);
      void parse_buderus(uint8_t * buf, size_t len);
  
  
      // Helper function (for better readability of code)
      void send_ACK_DLE();
      void send_NAK();
      void writeRequestValues();
      size_t genDataString(char* outbuf, uint8_t* inbuf, size_t len);
      void print_hex_buffer(uint8_t* buf, size_t len);
  
      uint32_t last_received_byte_time;
      Parser3964R parser;
      /** used to call the loop function of the sensors every x calls to the loop function of the component */
      uint8_t sensorLoopCounter;
    };
  
    } // namespace KM271
    } // namespace esphome

qschneider, 2022-12-13:
km271_params.h
  
    #pragma once
    
    #include <string>
    #include <vector>
    #include "esphome/core/component.h"
    #include "esphome/components/sensor/sensor.h"
    #include "esphome/components/binary_sensor/binary_sensor.h"
    #include "esphome/components/switch/switch.h"
    #include "esphome/components/number/number.h"
    #include <unordered_map>
    
    namespace esphome {
    namespace KM271 {
    
    
    enum Buderus_R2017_ParameterId {
      CFG_WW_Temperatur = 0x007e,
      CFG_WW_Aufbereitung = 0x0085,
      BW1HK1    = 0x8000, //: "Betriebswerte 1 HK1"
      BW2HK1    = 0x8001, //: "Betriebswerte 2 HK1"
      VSTHK1    = 0x8002, //: "Vorlaufsolltemperatur HK1"       (Grad)
      VITHK1    = 0x8003, //: "Vorlaufisttemperatur HK1"        (Grad)
      RSTHK1    = 0x8004, //: "Raumsolltemperatur HK1"          (Grad)
      RITHK1    = 0x8005, //: "Raumisttemperatur HK1"           (Grad)   
      EOZHK1    = 0x8006, //: "Einschaltoptimierungszeit HK1"
      AOZHK1    = 0x8007, //: "Ausschaltoptimierungszeit HK1"
      PLHK1     = 0x8008, //: "Pumpenleistung HK1"              (Prozent)
      MSHK1     = 0x8009, //: "Mischerstellung HK1"             (Prozent)
      NB01      = 0x800a, //: "nicht belegt"
      NB02      = 0x800b, //: "nicht belegt"
      KLHK1_P10 = 0x800c, //: "Heizkennlinie HK1 bei + 10 Grad" (Grad)
      KLHK1_P00 = 0x800d, //: "Heizkennlinie HK1 bei 0 Grad"    (Grad)
      KLHK1_N10 = 0x800e, //: "Heizkennlinie HK1 bei - 10 Grad" (Grad)
      NB03      = 0x800f, //: "nicht belegt"
      NB04      = 0x8010, //: "nicht belegt"
      NB05      = 0x8011, //: "nicht belegt"
  
      BW1HK2    = 0x8112, //: "Betriebswerte 1 HK2"
      BW2HK2    = 0x8113, //: "Betriebswerte 2 HK2"
      VSTHK2    = 0x8114, //: "Vorlaufsolltemperatur HK2"       (Grad)
      VITHK2    = 0x8115, //: "Vorlaufisttemperatur HK2"        (Grad)
      RSTHK2    = 0x8116, //: "Raumsolltemperatur HK2"          (Grad)
      RITHK2    = 0x8117, //: "Raumisttemperatur HK2"           (Grad)
      EOZHK2    = 0x8118, //: "Einschaltoptimierungszeit HK2"
      AOZHK2    = 0x8119, //: "Ausschaltoptimierungszeit HK2"
      PLHK2     = 0x811a, //: "Pumpenleistung HK2"              (Prozent)
      MSHK2     = 0x811b, //: "Mischerstellung HK2"             (Prozent)
      NB06      = 0x811c, //: "nicht belegt"
      NB07      = 0x811d, //: "nicht belegt"
      KLHK2_P10 = 0x811e, //: "Heizkennlinie HK2 bei + 10 Grad" (Grad)
      KLHK2_P00 = 0x811f, //: "Heizkennlinie HK2 bei 0 Grad"    (Grad)
      KLHK2_N10 = 0x8120, //: "Heizkennlinie HK2 bei - 10 Grad" (Grad)
      NB08      = 0x8121, //: "nicht belegt"
      NB09      = 0x8122, //: "nicht belegt"
      NB10      = 0x8123, //: "nicht belegt"
  
      BW1WW     = 0x8424, //: "Betriebswerte 1 WW"
      BW2WW     = 0x8425, //: "Betriebswerte 2 WW"
      WWST      = 0x8426, //: "Warmwassersolltemperatur"        (Grad)
      WWIT      = 0x8427, //: "Warmwasseristtemperatur",        (Grad)
      OZWW      = 0x8428, //: "Warmwasseroptimierungszeit"
      LPWW      = 0x8429, //: "Ladepumpe"                       ["aus", "Ladepumpe", "Warmwasserpumpe", "beide"]
  
      KVST      = 0x882a, //: "Kesselvorlaufsolltemperatur"     (Grad)
      KVIT      = 0x882b, //: "Kesselvorlaufisttemperatur"      (Grad)
      BET       = 0x882c, //: "Brennereinschalttemperatur"      (Grad)
      BAT       = 0x882d, //: "Brennerausschalttemperatur"      (Grad)
      KINT1     = 0x882e, //: "Kesselintegral 1"
[... 264 lines cut ...]
      {BLZ1S0, false, SensorType::UNSIGNED_INT, 0, "Brennerlaufzeit 1 Stunden 0", "h"},
      {BLZ2S2, false, SensorType::UNSIGNED_INT, 0, "Brennerlaufzeit 2 Stunden 2", "h"},
      {BLZ2S1, false, SensorType::UNSIGNED_INT, 0, "Brennerlaufzeit 2 Stunden 1", "h"},
      {BLZ2S0, false, SensorType::UNSIGNED_INT, 0, "Brennerlaufzeit 2 Stunden 0", "h"},
  
      {AT, false, SensorType::SIGNED_INT, 0, "Aussentemperatur", "°C"},             // (Grad)
      {ATD, false, SensorType::SIGNED_INT, 0, "Gedaempfte Aussentemperatur", "°C"}, // (Grad)
      {VVK, false, SensorType::UNSIGNED_INT, 0, "Versionsnummer VK", ""},
      {VNK, false, SensorType::UNSIGNED_INT, 0, "Versionsnummer NK", ""},
      {MODKENN, false, SensorType::UNSIGNED_INT, 0, "Modulkennung", ""},
      {NB12, false, SensorType::NONE, 0, "nicht belegt", ""},
      // Alarmstatus
      //{ALARM, false, SensorType::UNSIGNED_INT, 0, "Alarmstatus", ""}
      {ALARM, false, SensorType::BIT_AT_OFFSET, 0, "Alarm Abgasfuehler", ""},
      //{ALARM, false, SensorType::BIT_AT_OFFSET, 1, "Alarm 02", ""},
      {ALARM, false, SensorType::BIT_AT_OFFSET, 2, "Alarm Kesselvorlauffuehler", ""},
      //{ALARM, false, SensorType::BIT_AT_OFFSET, 3, "Alarm 08", ""},
      {ALARM, false, SensorType::BIT_AT_OFFSET, 4, "Alarm Brenner", ""},
      //{ALARM, false, SensorType::BIT_AT_OFFSET, 5, "Alarm 20", ""},
      {ALARM, false, SensorType::BIT_AT_OFFSET, 6, "Alarm HK2-Vorlauffuehler", ""},
      //{ALARM, false, SensorType::BIT_AT_OFFSET, 7, "Alarm 80", ""},
    
    };
    }
    }

qschneider, 2022-12-13:
i also streamlined hot_water_XX and warm_water_XX to ww_XX because its more consistant and easier to read.
This can also be done for heating_circuit_1 and heating_circuit_2 alias hc1 and hc2.
If you rather have it like it was, feel free to change it back.
If its ok, it also needs a change in the esphome yaml - so this would be a breaking change in configuration...

Bascht74, 2022-12-14:
Will you make a pull request?
I was planning to do exactly the same on the next weekend! Glad to see it already done -:)

@jensgraef @the78mole What do you two think about the streamlining suggestions?

jensgraef, 2022-12-14:
@Bascht74 I'm always a fan of making thinks more consistent and thus simpler.
I'll create a pull request on behalf of @qschneider

jensgraef, 2022-12-14:
But on the other hand - I'm not a fan of abbreviations - they tend to make things more complicated. So I'd like warm_water to stay warm_water and not become ww. But that's not a hill I'm prepared to die on :)

jensgraef, 2022-12-14:
I've created a pull request based on the suggested changes: #30

jensgraef, 2022-12-15:
@the78mole @qschneider @Bascht74: Do you think we can close this issue now?

qschneider, 2022-12-15:
maybe we have to give the entities more "self speaking" names in the future, but for now i think it's fine.

Bascht74, 2022-12-15:
Yes! Great!

closed 2022-12-16 by jensgraef