the78mole/esphome_components #11: handling of unsigned values

qschneider, 2022-12-06:
MSHK1, MSHK2 and all temperatures that could have an negative value should be treated as unsigned values not as INT to get usable values.

jensgraef, 2022-12-06:
Hi @qschneider - thanks for the hint - we were already thinking that there must be a way for those values to get negative.
I'll change the parsing for those.

jensgraef, 2022-12-07:
I've opened pull request #15 to address this. @qschneider Would you mind having a look whether I found all relevant values?

qschneider, 2022-12-07:
do you think it's necessary to define more sensors besides AT, ATD, MSHK1 and MSHK2 as signed integers?
Fi ABTMP will never reach negative values and some others like BET or BAT might not get greater than 127, but that doesn't mean their values are processed in the twos complement. Besides that at least MSHK2 works as expected.

jensgraef, 2022-12-07:
I added a commit - now only AT, ATD, MSHK1 and MSHK2 are interpreted as signed integers.

PR #15 (2022-12-09): switch reading of mixer position and temperature values to support fo…
…r signed values

This should fix #11 u
I have currently only tested it using positive values - as it's not that cold yet.
closed 2022-12-09 by the78mole