the78mole/esphome_components #12: device class of mixer_position

qschneider, 2022-12-06:
heating_circuit_1_mixer_position and heating_circuit_2_mixer_position have the device_class=DEVICE_CLASS_TEMPERATURE.
I'am not sure which device_class would be appropriate, or if it could be omitted if nothing matches.

https://www.home-assistant.io/integrations/sensor/

Bascht74, 2022-12-07:
mixer and pump shouldn't have that device class. 
They could have "power factor" (in %) or no device_class at all. For me they should have no device class at all.

jensgraef, 2022-12-09:
I agree.
@qschneider / @Bascht74 : Would one you you like to create a pull request for that? The device classes are configured in sensor.py

Bascht74, 2022-12-09:
@jensgraef I just made a pull request for that and some more little changes to sensors and binary sensors (e.g add the entity category).

If you make an example how to add the entity category for switch and number I could add them there, too.

Bascht74, 2022-12-09:
@qschneider Can you check?

qschneider, 2022-12-10:
Hi, seems to me everything works fine, except the seperation between entity_category=ENTITY_CATEGORY_DIAGNOSTIC and entity_category=ENTITY_CATEGORY_NONE. To me (personally) i wouldn't expect values in the diagnostic section, rather they belong in the sensors section.  
[image]
[image]

Bascht74, 2022-12-10:
For me diagnostic entities are the ones that are not primary and/or don't do measurements itself. You just use them to see whether your system is working in the right way.

according to: 
https://developers.home-assistant.io/docs/core/entity/

`Classification of a non-primary entity. 
Set to config for an entity which allows `changing the configuration of a device`, for example a switch entity making it possible to turn the background illumination of a switch on and off. 
Set to diagnostic for an entity `exposing some configuration parameter or diagnostics of a device but does not allow changing it`, for example a sensor showing RSSI or MAC-address.`

so the number and select entities should move to config entities. 

At the long end the primary entities should stay at `none`.  
E.g. main temperatures that are measured and climate entities for WW, HK1 and HK2.

So if you are distracted by splitting the sensors we should move more of them to the diagnostic entities section as they are not "primary". 
E.g. you don't get in touch with the Kesselvorlaufisttemperatur directly and the Brennerausschalttemperatur and Brennereinschalttemperatur, they are just diagnostic data.
On the other side, the WarmwasserIstTemperatur is a primary entity as the water come out of your tab with that temperature...

Besides that you normally add entities to your dashboards. you usually don't work that much with the device section in HA.

qschneider, 2022-12-10:
Mhh, it all is how you interpret
"Set to diagnostic for an entity exposing some configuration parameter or diagnostics of a device but does not allow changing it, for example a sensor showing RSSI or MAC-address"

So i would interpret it that way - put data in the diagnostic section which are related to the HW device itself, km271 or 2107. All others are sensor data that may change after user intervention, by time schedule or by the heater itself.

But i'am not keen on it, so it could stay that way unless someone else gets confused.

Bascht74, 2022-12-10:
OK, then we should leave it by now until we have more entities (e.g. more binary entities, sensor entities and control entities). 

For me the main output of the Buderus is warm water that has a specific temperature and Heating Water that has a specific temperature and a pump to get it to the radiators. These are the primary entities. 

All other are diagnostic and are used by km271/2107-system to control the system to produce these two primary entities.

Bascht74, 2022-12-10:
@jensgraef @the78mole you can close the issue.

closed 2022-12-10 by the78mole