the78mole/esphome_components #14: Something goes wrong with Tag/Nacht/Auto select and switch

the78mole, 2022-12-06:
When
https://github.com/the78mole/esphome_components/blob/1e22700df003cdebacc8df02c068563535ee4a12/components/km271_wifi/km271_params.cpp#L89

is executed by the param 0x0085 (short after initialization) , it shows quite wired trash...

[image]

I doubt, that data does not contain chars... 

When I change the switch, everything seems fine (ON, OFF, no trash).

[image]

Need to digg into it a bit deeper...

jensgraef, 2022-12-06:
I think it's wrong to print it as a string - that log line is something left over from some attempts before.
The configuration value is encoded as 0/1/2.
Is there something wrong beyond log line?

the78mole, 2022-12-06:
I didn't setup the select yet, so I can't tell if it's OK. But the switch seems to work as expected.

the78mole, 2022-12-10:
When using the select, I do not see the wired line of debug anymore, even if the debugging output line of code still exists. I'll try to change it, to get a reasonable output and create some PR.

qschneider, 2022-12-10:
Do you still get the weird values?

the78mole, 2022-12-10:
No, because I do not use the switch anymore. I use the select with lambda. But in fact, the line is wrong and if somebody uses the switch, he/she will get the crazy stuff an possibly the module will panic and reboot (infinitely). I created a PR :-)

the78mole, 2022-12-10:
#23 is handling this...

closed 2022-12-10 by the78mole