the78mole/esphome_components #17: Strange temperature-readings

jk-andersen, 2022-12-08:
Hi,

first of all: This is awesome - All of you guys participating and really printing boards and software for a 15+years heating-control. I absolutely adore this.

I got the board from Daniel and played around with it a bit and get some strange temperature-readings.

[image]

[image]

Temperature-Readings are junping 255 to 1 or 2 Degrees
I am not a pro in the depths of this code but maybe this has to do with my heating system not beeing oil, but a Gas-Heating "Logano G144" ? I flashed the newest version of the git today.

Any idea on your side?
Cheers,
Jan

jensgraef, 2022-12-08:
Hi @jk-andersen and welcome to the project! The temperature reading as it is at the moment can't deal with negative values. I didn't notice it yet as it was not that cold here ;-)
I've created a pull request to fix it here: https://github.com/the78mole/esphome_components/pull/15
So if you like, you can use the code from there branch or wait until it's on the main branch - which should be tomorrow or the day after.

jk-andersen, 2022-12-09:
Hi there,

I recompiled and flashed now after the merge but still get the same results, also for #18 
I am new to the ESP-Flashing game but I think I did it right as I was able tom add some new sensors.
I got a compile error though in between:

```
Compiling /data/buderus-km271/.pioenvs/buderus-km271/FrameworkArduino/esp32-hal-spi.c.o
/data/cache/platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-spi.c: In function 'spiTransferBytesNL':
/data/cache/platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-spi.c:922:39: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
                 uint8_t * last_out8 = &result[c_longs-1];
                                       ^
/data/cache/platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-spi.c:923:40: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
                 uint8_t * last_data8 = &last_data;
                                        ^
Compiling /data/buderus-km271/.pioenvs/buderus-km271/FrameworkArduino/esp32-hal-time.c.o
```
[logs_update-all.txt](https://github.com/the78mole/esphome_components/files/10197580/logs_update-all.txt)

jensgraef, 2022-12-09:
Mh, that's strange - with the merge to main both issues should be fixed.
The compile message you see is a warning that's originating in code from the ESPHome base code. It's not pretty but it's also nothing to worry about and should not affect our issue here.

I checked the ESPHome manual: If not specified differently, ESPHome only checks once after 1 day whether a github repository has changed: https://esphome.io/components/external_components.html#refresh -Maybe that's the issue here.
You could try adding ```refresh:3600s``` to the configuration of the external component - or wait some time - and see whether this helps.

One other thing: Strange readings for "Abgastemperatur" are to be expected if your heater does not have a hardware sensor for the exhaust temperature installed. My heater for example, doesn't have one.

Bascht74, 2022-12-09:
I add `refresh: 0s` 
... so every thime it is compiling it refreshes the sources...
:-)

closed 2022-12-09 by jensgraef