the78mole/esphome_components #18: esphome device "Failed to set up" in home assistant

cserem, 2022-12-09:
Hello!

When addig the ESPHOME device in HA it works as expected, but if I restart HA (not touching the km271 device at all), it fails to initialize/set up the device. I can see the device logs in ESPHOME, and can control the unit, only the HA connection is broken:
```
Error setting up entry km271 for esphome
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 365, in async_setup
    result = await component.async_setup_entry(hass, self)
  File "/usr/src/homeassistant/homeassistant/components/esphome/__init__.py", line 280, in async_setup_entry
    infos, services = await entry_data.async_load_from_store()
  File "/usr/src/homeassistant/homeassistant/components/esphome/entry_data.py", line 228, in async_load_from_store
    infos.append(cls.from_dict(info))
  File "/usr/local/lib/python3.10/site-packages/aioesphomeapi/model.py", line 77, in from_dict
    return cls(**init_args)
  File "<string>", line 15, in __init__
  File "/usr/local/lib/python3.10/site-packages/aioesphomeapi/model.py", line 63, in __post_init__
    super().__setattr__(field_.name, convert(val))
  File "/usr/local/lib/python3.10/site-packages/aioesphomeapi/util.py", line 46, in fix_float_single_double_conversion
    if value == 0 or not math.isfinite(value):
TypeError: must be real number, not NoneType
```

After some trial and error I was able to pinpoint the offender:
```
number:
  - platform: km271_wifi
    warm_water_temperature:
      name: "Warmwassersolltemperatur Tag"
```
If I uncomment this entity in the esphome yaml it works across HA restarts. (first the device needs to be re-added in HA, just removing  the number entity in HA isn't enough)

Could you please look at this?

Thanks

jk-andersen, 2022-12-09:
Hi,
just updated HA and can replicate the error.

`Logger: homeassistant.config_entries
Source: components/esphome/entry_data.py:247
First occurred: 09:40:33 (1 occurrences)
Last logged: 09:40:33

Error setting up entry buderus-km271 for esphome
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 372, in async_setup
    result = await component.async_setup_entry(hass, self)
  File "/usr/src/homeassistant/homeassistant/components/esphome/__init__.py", line 316, in async_setup_entry
    infos, services = await entry_data.async_load_from_store()
  File "/usr/src/homeassistant/homeassistant/components/esphome/entry_data.py", line 247, in async_load_from_store
    infos.append(cls.from_dict(info))
  File "/usr/local/lib/python3.10/site-packages/aioesphomeapi/model.py", line 83, in from_dict
    return cls(**init_args)
  File "<string>", line 16, in __init__
  File "/usr/local/lib/python3.10/site-packages/aioesphomeapi/model.py", line 69, in __post_init__
    super().__setattr__(field_.name, convert(val))
  File "/usr/local/lib/python3.10/site-packages/aioesphomeapi/util.py", line 46, in fix_float_single_double_conversion
    if value == 0 or not math.isfinite(value):
TypeError: must be real number, not NoneType`

jensgraef, 2022-12-09:
@cserem Thanks for reporting it and narrowing it down to the part that causes the trouble. The code is still so new I never restarted HA in the meantime.
Thanks to @jk-andersen for confirming it happens on multiple installations. I'll have a look into it - that's not show it should work ;-)

jensgraef, 2022-12-09:
I've created pull request #19 to fix this issue.
If you like you can use the branch   ```fix-home-assistant-startup-error-when-using-warm-water-number``` and see if the issue is gone.

PR #19 (2022-12-09): fix startup error on home assistant if warm water number field is used
add step parameter to number. if it is missing the esp home integration will raise an exception on startup.

This fixes #18
closed 2022-12-09 by the78mole