the78mole/esphome_components #29: Add device class temperature to warm_water_temperature

the78mole, 2022-12-14:
Since release ESPhome release 2022.12.0, adding DEVICE_CLASS to number inputs is possible.

See: [esphome#4042](https://github.com/esphome/esphome/pull/4042)

But since we would force users to upgrade to latest ESPhome release immediately, we will delay this for a few weeks.

Bascht74, 2023-01-12:
As 2023.01.0 around the corner I think this could by implemented now...
What do you think?

the78mole, 2023-01-12:
Probably... :-)

qschneider, 2023-01-14:
If we declare min_version in our ESPhome.yaml, we could force a certain ESPhome version. That way we can introduce features that require a certain ESPhome version. What do think about it ?

https://esphome.io/components/esphome.html#minimum-esphome-version

the78mole, 2023-01-14:
@qschneider Did you already test it?

qschneider, 2023-01-14:
the code snippet in buderus-km271-writable.yaml ?
Yes, this is from my actual config file.
[buderus-km271-writable.yml.zip](https://github.com/the78mole/esphome_components/files/10417428/buderus-km271-writable.yml.zip)

jensgraef, 2023-05-22:
2022.12.0 is now more than 5 months old. Does someone want to add the device class now?

the78mole, 2024-04-29:
Just stubled over this old open issue 😂
Shall we add this to the default YAML now?

closed 2025-01-24 by the78mole