the78mole/esphome_components #49: UNSIGNED_INT_DIVIDED_BY_2 not handling .5°C values

qschneider, 2023-01-04:
I found our, that rather than just dividing those values by 2, they seem to have the purpose to display 0.5 step temperatures as a float variable.

This should be the case for   
    heating_circuit_1_room_target_temperature, RSTHK1
    heating_circuit_1_room_temperature, RITHK1
    heating_circuit_2_room_target_temperature, RSTHK2
    heating_circuit_2_room_temperature, RITHK2

at the moment I have also an issue with heating_circuit_2_room_temperature as it is displaying 55,0°C but I shouldn't.
maybe this is related ?

qschneider, 2023-01-04:
maybe the code is working fine because i found this in my log.

    [21:12:06][D][km271:036]: Parameter 0x8117: (Data: 1, 0x6E )
    [21:12:06][D][sensor:127]: 'HK2 Raumisttemperatur': Sending state 55.00000 °C with 1 decimals of accuracy

still it's unclear to me how to handle this value, because no room thermostat is attached...

jensgraef, 2023-01-06:
Could you clarify on that issue regarding the code not handling values ending in .5°C?
It seems to be working fine at my installation:
[image]

closed 2023-01-07 by qschneider