the78mole/esphome_components #53: What does the following formula do?

Bascht74, 2023-01-14:
@qschneider What does the following formula exactly do?

```
  - service: number.set_value
    data:
      value: >-
        {{ ([55.0, (10.5-(states('sensor.gedaempfte_aussentemperatur') |
        float))*2+45, 45.0]|sort)[1] }}
```

Thx for a hint!

qschneider, 2023-01-14:
It calculates warm water temperature off sensor.gedaempfte_aussentemperatur if its in the range between 45°C and 55°C. Below or above that range ist returns a fixed value of 45°C or 55°C - see example below.
For that it used sort()[1] to always return the middle number of (45, 55, VAL).

    gAT -> WW Soll
    > 11  -> 45
    10-> 46
    9 -> 48
    8 -> 50
    7 -> 52
    6 -> 54
    < 6 -> 55

It suits my need to have a moving warm water temperature in that range to optimize heater starts. It maybe just an edge case depending on your heating setup.

Does this answers your question ?

qschneider, 2023-01-14:
To discuss features, automations, snippets and more it's also possible to use the discussions tab of this repo.

qschneider, 2023-01-17:
If you have no further questions, we might close here ?

closed 2023-01-17 by jensgraef