the78mole/esphome_components #62: WiFi Authentication not working

christian-heusel, 2023-03-11:
My Dad and I recently bought and installed a [Buderus KM271 WiFi Replacement](https://www.tindie.com/products/the78mole/buderus-km271-wifi-replacement) on Tindie and it seems like the initial WiFi login to the "Fallback Hotspot" is not working. As described in the blogpost we used the password `Z8zfajgxVvNw` and waited far more than a minute (see kernel log). 

We tried a Fairphone 4 and my laptop.

Any Idea what could be going wrong?

<details>
 <summary>Click here to show <code>dmesg</code> output</summary>

```
[  648.689817] wlp3s0: send auth to 1c:9d:c2:f3:ba:6d (try 1/3)
[  648.695741] wlp3s0: send auth to 1c:9d:c2:f3:ba:6d (try 2/3)
[  648.702465] wlp3s0: send auth to 1c:9d:c2:f3:ba:6d (try 3/3)
[  648.708755] wlp3s0: authentication with 1c:9d:c2:f3:ba:6d timed out
[  652.512696] wlp3s0: authenticate with 1c:9d:c2:f3:ba:6d
[  652.986777] wlp3s0: send auth to 1c:9d:c2:f3:ba:6d (try 1/3)
[  653.008248] wlp3s0: send auth to 1c:9d:c2:f3:ba:6d (try 2/3)
[  653.030288] wlp3s0: send auth to 1c:9d:c2:f3:ba:6d (try 3/3)
[  653.057771] wlp3s0: authentication with 1c:9d:c2:f3:ba:6d timed out
[  677.017988] wlp3s0: authenticate with 70:a7:41:cb:64:59
[  677.460475] wlp3s0: send auth to 70:a7:41:cb:64:59 (try 1/3)
[  677.463840] wlp3s0: authenticated
[  677.465261] wlp3s0: associate with 70:a7:41:cb:64:59 (try 1/3)
[  677.470377] wlp3s0: RX AssocResp from 70:a7:41:cb:64:59 (capab=0x1011 status=0 aid=4)
[  677.470627] wlp3s0: associated
[  677.565791] IPv6: ADDRCONF(NETDEV_CHANGE): wlp3s0: link becomes ready
[  699.117961] wlp3s0: deauthenticating from 70:a7:41:cb:64:59 by local choice (Reason: 3=DEAUTH_LEAVING)
[  703.989764] wlp3s0: authenticate with 1c:9d:c2:f3:ba:6d
[  704.457251] wlp3s0: send auth to 1c:9d:c2:f3:ba:6d (try 1/3)
[  704.460620] wlp3s0: authenticated
[  704.462150] wlp3s0: associate with 1c:9d:c2:f3:ba:6d (try 1/3)
[  704.471124] wlp3s0: RX AssocResp from 1c:9d:c2:f3:ba:6d (capab=0x431 status=0 aid=1)
[  704.471434] wlp3s0: associated
[  704.529005] IPv6: ADDRCONF(NETDEV_CHANGE): wlp3s0: link becomes ready
[  706.728679] wlp3s0: Connection to AP 1c:9d:c2:f3:ba:6d lost
[  707.055689] rtw_8822ce 0000:03:00.0: timed out to flush queue 1
[  719.563669] wlp3s0: authenticate with 1c:9d:c2:f3:ba:6d
[  719.878942] wlp3s0: send auth to 1c:9d:c2:f3:ba:6d (try 1/3)
[  719.885652] wlp3s0: authenticated
[  719.888869] wlp3s0: associate with 1c:9d:c2:f3:ba:6d (try 1/3)
[  719.896471] wlp3s0: RX AssocResp from 1c:9d:c2:f3:ba:6d (capab=0x431 status=0 aid=1)
[  719.896780] wlp3s0: associated
[  722.185830] wlp3s0: Connection to AP 1c:9d:c2:f3:ba:6d lost
[  741.911740] wlp3s0: authenticate with 1c:9d:c2:f3:ba:6d
[  742.222525] wlp3s0: send auth to 1c:9d:c2:f3:ba:6d (try 1/3)
[  742.225918] wlp3s0: authenticated
[  742.228836] wlp3s0: associate with 1c:9d:c2:f3:ba:6d (try 1/3)
[  742.237025] wlp3s0: RX AssocResp from 1c:9d:c2:f3:ba:6d (capab=0x431 status=0 aid=1)
[  742.237322] wlp3s0: associated
[  742.319332] IPv6: ADDRCONF(NETDEV_CHANGE): wlp3s0: link becomes ready
[  759.210433] wlp3s0: deauthenticating from 1c:9d:c2:f3:ba:6d by local choice (Reason: 3=DEAUTH_LEAVING)
[  760.159254] wlp3s0: authenticate with 1c:9d:c2:f3:ba:6d
[  760.472416] wlp3s0: send auth to 1c:9d:c2:f3:ba:6d (try 1/3)
[  760.475815] wlp3s0: authenticated
[  760.478792] wlp3s0: associate with 1c:9d:c2:f3:ba:6d (try 1/3)
[  760.585665] wlp3s0: associate with 1c:9d:c2:f3:ba:6d (try 2/3)
[  760.692117] wlp3s0: associate with 1c:9d:c2:f3:ba:6d (try 3/3)
[  760.799019] wlp3s0: association with 1c:9d:c2:f3:ba:6d timed out
[  764.653394] wlp3s0: authenticate with 1c:9d:c2:f3:ba:6d
[  764.958850] wlp3s0: send auth to 1c:9d:c2:f3:ba:6d (try 1/3)
[  765.065730] wlp3s0: send auth to 1c:9d:c2:f3:ba:6d (try 2/3)
[  765.175471] wlp3s0: send auth to 1c:9d:c2:f3:ba:6d (try 3/3)
[  765.282134] wlp3s0: authentication with 1c:9d:c2:f3:ba:6d timed out
[  769.383220] wlp3s0: authenticate with 1c:9d:c2:f3:ba:6d
[  769.695754] wlp3s0: send auth to 1c:9d:c2:f3:ba:6d (try 1/3)
[  769.802361] wlp3s0: send auth to 1c:9d:c2:f3:ba:6d (try 2/3)
[  769.908803] wlp3s0: send auth to 1c:9d:c2:f3:ba:6d (try 3/3)
[  770.018971] wlp3s0: authentication with 1c:9d:c2:f3:ba:6d timed out
[  774.613017] wlp3s0: authenticate with 1c:9d:c2:f3:ba:6d
[  774.919000] wlp3s0: send auth to 1c:9d:c2:f3:ba:6d (try 1/3)
[... 343 lines cut ...]
[ 1548.437498] wlp3s0: send auth to 1c:9d:c2:f3:ba:6d (try 1/3)
[ 1548.441698] wlp3s0: authenticated
[ 1548.443876] wlp3s0: associate with 1c:9d:c2:f3:ba:6d (try 1/3)
[ 1548.550524] wlp3s0: associate with 1c:9d:c2:f3:ba:6d (try 2/3)
[ 1548.657388] wlp3s0: associate with 1c:9d:c2:f3:ba:6d (try 3/3)
[ 1548.764014] wlp3s0: association with 1c:9d:c2:f3:ba:6d timed out
[ 1562.367637] wlp3s0: authenticate with 1c:9d:c2:f3:ba:6d
[ 1562.677229] wlp3s0: send auth to 1c:9d:c2:f3:ba:6d (try 1/3)
[ 1562.783955] wlp3s0: send auth to 1c:9d:c2:f3:ba:6d (try 2/3)
[ 1562.890342] wlp3s0: send auth to 1c:9d:c2:f3:ba:6d (try 3/3)
[ 1562.997108] wlp3s0: authentication with 1c:9d:c2:f3:ba:6d timed out
[ 1575.920502] wlp3s0: authenticate with 70:a7:41:cb:64:5a
[ 1576.401812] wlp3s0: send auth to 70:a7:41:cb:64:5a (try 1/3)
[ 1576.405466] wlp3s0: authenticated
[ 1576.406592] wlp3s0: associate with 70:a7:41:cb:64:5a (try 1/3)
[ 1576.411796] wlp3s0: RX AssocResp from 70:a7:41:cb:64:5a (capab=0x1111 status=0 aid=8)
[ 1576.412214] wlp3s0: associated
[ 1576.470389] IPv6: ADDRCONF(NETDEV_CHANGE): wlp3s0: link becomes ready
```

</details>

---

cc @Heusel

qschneider, 2023-03-12:
Hi, I (we) changed in #55 the password in https://github.com/the78mole/esphome_components/blob/main/components/km271_wifi/km271-for-friends.yaml to "".
Please try and report back if this solved your problem so @the78mole could either edit his blogpost or change the pwd in that yaml back to the one posted in the blogpost. Cheers

the78mole, 2023-03-12:
Hi,
I most probably flashed the Firmware with the following settings (I can not remember, that this changed somewhen in the past):

```
wifi:

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Fallback Hotspot"
    password: "Z8zfajgxVvNw"
```

If nothing works, you should simply re-flash the firmware. If I shall support you, it would be of great benefit, if you describe me your actual setup and what is installed how (Home Assisstant, ESPhome,...).

Regards,
Daniel

christian-heusel, 2023-03-12:
Hey you two! 🤗

So the board was bought already flashed and there were no modifications made to the flashed image.

> I changed [...] the password [...] to "".

I cannot use an empty password in NetworkManager, but I also think that as the board came pre-flashed it does not contain an empty password but instead the one that @the78mole mentioned.

christian-heusel, 2023-03-12:
Closing this as we got it to work in the meantime! 😄 
The problem was insufficent power on the esp which we fixed by using a external power supply!

closed 2023-03-12 by christian-heusel