the78mole/esphome_components #63: How to translate?

man55, 2023-04-17:
Hi and thanks for a great device!

I wanted to ask what is the easiest way I can do the translation of the names of the sensors for myself, in order to do this only once for all? So that in the future, in the event of any events, updates, the need to add the device again, etc...  I did not have to translate again.

I already did it manually directly in Hjme Assistant. But I had to do it twice. And today I renamed the device and it turned out to be a complicated process with flashing, after which I had to remove and add the device to the HA again.
And now I need to rename all the sensors for the third time, and there are a lot of them and I'm tired).

p.s. And it would also be nice to translate the same mode options, such as: "Aus, Ein, Nacht, Tag, Abschalt, Reduziert, Raumhalt, Aussenhalt, Heizkörper ....", the last of which I could not understand at all.

the78mole, 2023-04-17:
Hi man55,

you just need to save your YAML file and reuse this for configuration. The one part is the identifier of the sensor in the C-Code, the other is the Name, HA gets for the sensor.

```
sensor:
  - platform: km271_wifi
    heating_circuit_1_flow_target_temperature:    # <- This is the HW sensor id (not changable) to identify the parameter
      name: "HK1 Vorlaufsolltemperatur"           # <- This is the Home Assistant name (transformed to an entity ID) and changeable
```

Hope this helps :-)

Regards,
Daniel

man55, 2023-04-18:
Thank you. So I already tried to do it. I saved the YAML to /config/esphhome and named the sensors there. As a result, I doubled all the sensors, that is, there were twice as many of them. As I understand it, it used two YAMLs, one is mine, and the second is specified by default in the km271-for-friends-xxxxxx.yaml itself, which is created automatically during the first initialization:
```
substitutions:
  name: "km271-for-friends-xxxxxx"
packages:
  the78mole.km271-wifi: github://the78mole/esphome_components/components/km271_wifi/km271-for-friends.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
api:
.........
```
Then I tried to change directly /config/esphome/.esphome/packages/040b63aa/components/km271_wifi/km271-for-friends.yaml, but this was also unsuccessful, on the contrary, I lost many sensors and I did not understand why.
In any case, both methods are unreliable, since everything disappears when reinitialized.

But I googled and found another way: I created a customize.yaml file, in which I named the sensors, then connected it to configuration.yaml. Now everything works correctly, since I did not set the sensors again, but simply assigned them friendly names.

This in configuration.yaml:
```
homeassistant:
  customize: !include customize.yaml
```

This in customize.yaml:
```
binary_sensor.alarm_abgasfuehler:
  friendly_name: 'Аварія датчика вихлопних газів'

binary_sensor.alarm_brenner:
  friendly_name: 'Аварія пальника'

number.heizkreis_1_raumtemperatur_offset:
  friendly_name: 'Зсув кімнатної температури'
.....

```
The main problem was to create a list of all entities, but I just copied them from the entities page

the78mole, 2023-04-18:
Yes, esphome bevaes a bit strange, when removing and renaming devices, but that it generates such a lot of problems is new to me. Fortunately it works for you now :-)

closed 2023-04-18 by the78mole