the78mole/esphome_components #64: select.sommer_winter_umschalttemperatur only changing heat circuit 1

AlexKusnezov, 2023-05-05:
Sensor select.sommer_winter_umschalttemperatur only chnaged heating circuit 1, HK2 stays unchanges as it was set by 2107. Also sensor binary_sensor.hk2_sommer_modus also stays on
I xan verify that on the 2107 wheb I switch between circuits. (summer-icon reappears on the display and hk2 won't jump on, stays in summer mode)

Can you take a look please?

jensgraef, 2023-05-06:
Hi Alex and welcome to our github page!
I had a short look at the code and looks like the 2107 has two select fields for that and that  select.sommer_winter_umschalttemperatur affects only one of it.
https://github.com/the78mole/esphome_components/blob/main/components/km271_wifi/km271_communication_components.cpp#L175

So I'd suggest changing the code to rename select.sommer_winter_umschalttemperatur to reflect that it only chances heating circuit 1. Then we can add code for another select to change hc2, too.
Unfortunately I'm unable to test that as I only have a heater with one circuit available.
@qschneider: Would you mind adding and testing the select for hc2?

https://github.com/the78mole/esphome_components/blob/main/components/km271_wifi/km271_communication_components.cpp#L175

dewenni, 2023-05-06:
I can confirm that "Sommer ab" and also "Frost ab" are separate values for HC1 and HC2.
The setvalue commands are different and also the status values are different.

I have also separated this some time ago in my project.
Maybe it is usefull to see the differences and the protocol informations
here the related commit: [dewenni/ESP_Buderus_KM271](https://github.com/dewenni/ESP_Buderus_KM271/commit/48dfd3312102e385f6efc4cbdf598884eff835c7)

AlexKusnezov, 2023-05-06:
@jensgraef thanks guys, I really appreciate the work you have done here. It's truly amazing to able to interface with 2107 in such a granular way.

qschneider, 2023-05-07:
@jensgraef i checked telegrams 08 00 65 09 65 65 65 65 (summer hc2 Temp = 9) and 08 31 65 65 65 65 65 04 (frost hc2 Temp=4). I can confirm that the summer mode needs to be setup separately if hc2 is present.

For the frost one it seems to be fine to just use 07 31 65 65 65 65 65 04 which will set the value while sending the hc2 telegramm result in no change or response at all.
Can someone with hc2 setup confirm this as well?

qschneider, 2023-05-07:
i made the changes to #65, pls review and commit, thx

jensgraef, 2023-05-09:
@AlexKusnezov: The change is now on master and you should find a new binary version here: https://github.com/the78mole/esphome_components/blob/gh-pages/README.md

Please close this issue if it's working as expected.

Cheers
Jens

AlexKusnezov, 2023-05-10:
@jensgraef thanks, what I did before was just pasted this yaml in my esphome: https://github.com/the78mole/esphome_components/blob/main/components/km271_wifi/km271-for-friends.yaml
the compiled and installed it via ota.
You think I can do this again? I'm not sure if the bin file has everything enabled. 
When I recieved the unit a lot of this, especially the HC2 stuff was read-only. Is it the same with the bin?

jensgraef, 2023-05-10:
@AlexKusnezov yes, you can also do it that way. It how I do it myself ;-)
Here's what need to be changed if you have made any customization in your yaml file and do not want to start from the one in the repository:
https://github.com/the78mole/esphome_components/commit/b7919eacc8d0ed6ebf944338ec7b0ba03c741531

AlexKusnezov, 2023-05-10:
@jensgraef thanks for the clarification. I kept everything pretty much vaniöla beside the wifi stuff. I'll try that tomorrow and report back.

AlexKusnezov, 2023-05-11:
@jensgraef everything went evry well. and it works really good. I can now set summer temp for both hc's indivudually.
Thank you so much!

I'm just wondering why you can't do this on the Logamatic unit it self, because setting the temp there will change tamp for both hc's

jensgraef, 2023-05-11:
That's good to hear!
As for why you can't do this via the the menu one the Logamatic: I don't have a clue. I'm sure somebody thought it a good idea ;-)

closed 2023-05-11 by jensgraef