the78mole/esphome_components #66: [Feature Request] Update the clock

mohnen, 2023-05-08:
Would it be possible to set the clock of the Logamatic via this interface?
Currently, I use BUDERUS Fernbedienung BFU for this.

jensgraef, 2023-05-08:
Hi mohnen and welcome to our little project!

It looks to me that it's possible to send a message to the Logamatic to update the time.
dewenni has implented this in his firmware: https://github.com/dewenni/ESP_Buderus_KM271/blob/main/src/km271.cpp#L1530

Currently this software can not do it - but I don't think it's too much work.
Do you have a suggestion on how to handle the Home Assistant side?
We could start with a simple button that takes the current time and date according to home assistant and write it to the heater.

mohnen, 2023-05-08:
Hi,

I guess a button would be nice. An automation which "presses" the button
once a day should cover the daylight savings.

Thanks
Markus

mohnen, 2023-05-09:
I just had an "add-on" idea: If it would be possible to also connect a HA temperature sensor to the "HK1 Raumisttemperatur" (which is currently a sensor, as it is set by the BFU) I could get rid of this BFU completely :)
At least as long as I can manually set "HK1 Absenkungsart" to "Raumhalt" even w/o the physical BFU connected.

the78mole, 2023-05-09:
You mean, if you could set the actual room temperature in the Buderus by Home Assistant? Hm... I don't know, if the Buderus Unit supports writing to this register on the bus we are using... Maybe @jensgraef can estimate it based on his experience.

jensgraef, 2023-05-09:
I don't think we can write to "HK1 Raumisttemperatur". And if it was possible I wouldn't know how to ;-)

What you can actually do to lose the BFUs writing values to
   
config_heating_circuit_1_room_target_temperature_night:
   name: "Heizkreis 1 Raumsolltemperatur Nacht"

config_heating_circuit_1_room_target_temperature_day:
    name: "Heizkreis 1 Raumsolltemperatur Tag"

Without the BFU the Logamatic is estimating the room temperature by some way unknown to be.
But if you raise    "Heizkreis 1 Raumsolltemperatur Nacht" it will heat more at night and if you raise  "Heizkreis 1 Raumsolltemperatur Tag" it will heat more during the day.
It think one could use this in combination with a HA room temperature sensor control the room temperature.

If you want to build this, be aware that every time you write a value to the logamatic it will save it to it's internal EEPROM.
Like an flash drive the EEPROM will only be able to handle a certain amount of write cycles before it breaks - I believe around one million for the one built into the logamatic.
So if you update the value every 10 seconds around the clock, you'll probably wear out the EEPROM within a year.
Writing once or twice every hour should  be safe a  this only leads to roughly 20.000 writes a year.

jensgraef, 2023-05-11:
> > Yes. Everything not stored in the EEPROM would be gone if you turn off the logamatic or have a power outage. The EEPROM is the only way for the device to "remember" values.
> 
> But why is the current value "HK1 Raumisttemperatur" something the device needs to remember? It is updated by the BFU continously (or at least in fixed intervals).

@mohnen I think we had a misunderstanding here. You are right, "HK1 Raumisttemperatur" is a value that is most certainly read directly from the BFU.
But I don't think we can write this value via the KM271_WIFI.
What is possible to achieve your goal:
* disable the BFU
* use the configuration setting "config_heating_circuit_1_room_target_temperature_day"(+night) to increase or decrease the temperature of the room using home assistant

These configuration settings are stored in the EEPROM.

jensgraef, 2023-05-21:
Hi @mohnen!

I just open pull request #68 to allow setting the date and time.
If you like you can test it using the branch feature/set-buderus-time - or wait until it's merged to master.

It's currently not really polished :)
The following link explains how to enable it.
https://github.com/the78mole/esphome_components/blob/1f9c4ef13fa7c7bce1227b91ff6b8945ffd1b058/components/km271_wifi/README.md#how-do-i-set-the-time--day-of-week-of-the-heater-via-home-assistant

margritstein, 2024-02-03:
@jensgraef Are you still planning of merging it into master or does @the78mole need to do it?

jensgraef, 2024-02-04:
> @jensgraef Are you still planning of merging it into master or does @the78mole need to do it?

I'd like to get a thumbs up from @the78mole before merging it. I've pinged him on the pull request.

the78mole, 2024-02-04:
Hi Jens, if you can merge it, feel free to do so. I hope it is tested enough :-P

jensgraef, 2024-02-04:
It has been merged to main.

closed 2024-02-04 by jensgraef