the78mole/esphome_components #67: [Update Error] esphome/components/select/select.h: No such file or directory

arnkit, 2023-05-12:
Hi Everyone,

I'm running Home Assitant in the newest version native on RasPi4. According to HA my current installed Version von ESPHome is 2022.12.2. Latest Version is 2023.4.4.
The KM271 is my only ESPHome device.
When I try to update I get the following error:

```
2023-05-12 13:10:46.869 ERROR (MainThread) [homeassistant.components.esphome.update] Error compiling heizung.yaml. Try again in ESPHome dashboard for error
2023-05-12 13:10:49.344 ERROR (MainThread) [homeassistant.components.esphome.update] Error OTA updating heizung.yaml. Try again in ESPHome dashboard for error
```

When I try to update in the ESPHome dashboard  it will give me a few further information. 

```
Updating /config/esphome/heizung.yaml
------------------------------------------------------------

INFO Reading configuration /config/esphome/heizung.yaml...
WARNING GPIO2 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO4 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Generating C++ source...
INFO Compiling app...
Processing heizung (board: esp32dev; framework: arduino; platform: platformio/espressif32 @ 5.3.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- AsyncTCP-esphome @ 1.2.2
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 2.1.0
|   |-- AsyncTCP-esphome @ 1.2.2
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
Compiling /data/heizung/.pioenvs/heizung/src/esphome/components/km271_wifi/km271.cpp.o
Compiling /data/heizung/.pioenvs/heizung/src/esphome/components/km271_wifi/km271_communication_components.cpp.o
Compiling /data/heizung/.pioenvs/heizung/src/esphome/components/km271_wifi/km271_helpers.cpp.o
Compiling /data/heizung/.pioenvs/heizung/src/esphome/components/km271_wifi/km271_params.cpp.o
In file included from src/esphome/components/km271_wifi/km271.h:10,
                 from src/esphome/components/km271_wifi/km271.cpp:1:
src/esphome/components/km271_wifi/km271_communication_components.h:7:10: fatal error: esphome/components/select/select.h: No such file or directory
 #include "esphome/components/select/select.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
In file included from src/esphome/components/km271_wifi/km271_communication_components.cpp:1:
src/esphome/components/km271_wifi/km271_communication_components.h:7:10: fatal error: esphome/components/select/select.h: No such file or directory
 #include "esphome/components/select/select.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
In file included from src/esphome/components/km271_wifi/km271_params.cpp:6:
src/esphome/components/km271_wifi/km271_communication_components.h:7:10: fatal error: esphome/components/select/select.h: No such file or directory
 #include "esphome/components/select/select.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [/data/heizung/.pioenvs/heizung/src/esphome/components/km271_wifi/km271_params.cpp.o] Error 1
*** [/data/heizung/.pioenvs/heizung/src/esphome/components/km271_wifi/km271_communication_components.cpp.o] Error 1
*** [/data/heizung/.pioenvs/heizung/src/esphome/components/km271_wifi/km271.cpp.o] Error 1
========================== [FAILED] Took 9.10 seconds ==========================
=========== [ERROR] /config/esphome/heizung.yaml ===========

======================== [SUMMARY] ========================
  - /config/esphome/heizung.yaml: FAILED
```
 
Any ideas?

the78mole, 2023-05-12:
Better, if you would have pasted your YAML also...
Nevertheless, yes, I have an idea. 
Please keep all sections in your yaml, even if they are empty:

```
[...]
binary_sensor:
number:
select:
sensor:
text_sensor:
```

jensgraef, 2023-05-21:
Hi and welcome @arnkit !
Did the suggestion by @the78mole help?

arnkit, 2023-05-24:
HI @the78mole and @jensgraef,

Thanks for your ideas. I was indeed missing the following sections:

```
select:
text_sensor:
```

Once they've been added to the yaml.file compilation and update went throught like charme.

Thanks again!

closed 2023-05-24 by arnkit