the78mole/esphome_components #69: Updating ESPHome Itself

AlexKusnezov, 2023-06-02:
Hey guys,

I'm kind of new to ESPHome and wanted to ask how it is with the updates of the framework itself in relation to this firmware.

Currently I'm on 2023.4.4 and want to g update to 2023.5.5. Is it safe to do that by just hitting the upadte button?
[image]
Or does it have to be some kind of apporved by you guys first?

Sorry for being overconcerned here ;)

the78mole, 2023-06-02:
Should not be a problem. I can not guarantee, that the APIs we use keep untouched, but up to now, we never had issues. 
We also try to not take over every cool new feature of new base firmwares, to not break our component for people still using not up-to-date ESPhome addon.

AlexKusnezov, 2023-07-29:
hey just wanted to report, that meanwhile I've successfully updated to Esphome 2023.7.0 and everything is good.
[image]

AlexKusnezov, 2024-01-25:
2023.12.9 and still going:
[image]

closed 2024-04-29 by the78mole