the78mole/esphome_components #75: HK2 Raumtemperatur Offset shows 124,5 after ESPHome update

EPGK, 2024-06-05:
Hi!
After updating ESPHome the Offset of HK2 Raumtemperatur shows 124,5 instead off -4.
As far as I can tell, it's not changing anything. Just reading the value wrong.
[image]
[image]

the78mole, 2024-06-05:
To be honest, I don't understand, why you experience this issue. I can not find any evidence, that this parameter has changed in the code and it still looks very similar to the HK1 offset. Unfortunately, I have no HK2 installed and therefore, no experience with it. Do you see similar problems also with HK1?

EPGK, 2024-06-05:
I just redid the ESPHome update with HK1 Offset -1 and it jumpted to plus 5
[image]
The same seems to happen with HK2. Its 5 but shows 124,5
[image]

the78mole, 2024-06-05:
When I change my HK1 room temp offset to 2°C in the UI, I see the following LOG:
[image]

Explained: 04h = 04d => 4 / 2 = 2.0

But the parameter itself has in fact not a 0.5°C accuracy and also on the Buderus itself, I can only change the value in steps of 1 °C. It is only represented by such parameter.

The code for HK1 tells the correct parameter and the correct interpretation:
```C
{config_heating_circuit_1_room_temperature_offset, CFG031, true, SensorType::BYTE_DIVIDED_BY_2_AT_OFFSET, 3, "CFG_HK1_Raumtemperatur Offset", "°C"}
```

The (new) part in my ESPhome config looks like the following, I have just added to dig into this problem:
```yaml
number:
  - platform: km271_wifi
    config_ww_temperature:
      name: "Warmwassersolltemperatur Tag"
      mode: slider
    config_heating_circuit_1_design_temperature:
      name: "Auslegungstemperatur"
      mode: slider
    config_heating_circuit_1_room_target_temperature_day:
      name: "HK1 Raumsolltemperatur Tag"
    config_heating_circuit_1_room_temperature_offset:
      name: "HK1 Room Temp Offset"
```

Could you please check, if everything aligns with my stuff and add the Log output of the ESP (like in my first image), where the raw values of the received params are printed? 

Regards,
Daniel

EPGK, 2024-06-05:
Hi Daniel,
I'm not close to the UI at the moment.
But logs from the GUI all look fine.
[image]
I think that the discrepancy only appears with negativ values.
When I update the ESPHome with positiv values, these stay like before the update.
Only the negativ values behave strange when updating the ESPHome.
Regards,
Erik

the78mole, 2024-06-05:
OK, will check with the new ESPhome (right updating my Home Assistant). I'll report back tomorrow.

the78mole, 2024-06-06:
OK, now updated and tested the negative values and it really behaves strage. I'm not sure, if it is related to the ESPhome update or simply a bug in my code... But it will take some time to fix it... (not today)

jensgraef, 2024-06-07:
Hi guys, I'll try to have a look into it this weekend.

jensgraef, 2024-06-09:
@the78mole you were exactly right - we need SIGNED_BYTE_DIVIDED_BY_2_AT_OFFSET to deal with negative values for the offset temperatures. I added that in #76.
Furthermore I fixed a rounding error that was setting the offset to -1.5 if you choose -2 in home assistant.

The Buderus control unit internally works with steps of 0.5 for this - you can see the control target temperatures changing each time you increment / decrement by 0.5.

@EPGK Thanks for the bug report! If you like you can test whether the changes I made on the branch completely solve this issue.
For that youl'll need to change the line
```
  - source: github://the78mole/esphome_components@main
```
to 
```
  - source: github://the78mole/esphome_components@bugfix/fix-error-handling-negative-temperature-offsets
```
in the YAML file and rebuild.
Or you can wait until we have it merged to main :)

the78mole, 2024-06-09:
> Or you can wait until we have it merged to main :)

Which I just did :-P

EPGK, 2024-06-10:
Hi!
I've updated ESPHome twice to be sure everything stays as it should.
Works like a charm.
Thanks for all the work and effort.

the78mole, 2024-06-10:
You are welcome! Many thanks for your report and also many thanks to @jensgraef for quickly correcting it. I love our tiny KM271 community 🥰

closed 2024-06-10 by the78mole