the78mole/esphome_components #77: Error Update ESPHome 2024.6.1

hergie1976, 2024-06-20:
I get the following error in ESPHome 2024.6.1 when i update KM wirelessly

INFO ESPHome 2024.6.1
INFO Reading configuration /config/esphome/buderus-heizung.yaml...
INFO Updating https://github.com/the78mole/esphome_components.git@main
WARNING GPIO2 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
Failed config

At least one platform must be specified for 'ota'; add 'platform: esphome' for original OTA functionality

Markus1971a, 2024-06-22:
Unfortunately I have the same problem, but also since version 2024.6.0

hergie1976, 2024-06-22:
Hi,

you have to change the config.

Insert the following part in the config yaml. Please use the editor of ESPHome!

```
ota:
  platform: esphome
```

Best Regrads

Heiko

the78mole, 2024-06-22:
Thanks Heiko 🤗

closed 2024-06-22 by the78mole