the78mole/esphome_components #78: ESPHome Update Struggle

AlexKusnezov, 2024-06-28:
Hey guys,

since 2024.6 is out now I wanted to give it a shot and update my module.
So I went and adopted the config to:
```
ota:
  - platform: esphome
    password: "some-ota-pw" 
```

However, since 2024.5.0 (last succefull update) it would compile, then try to upload and the resulting in:
` [Errno 104] Connection reset by peer`
with the module just restart and sty on 2024.5.0

I'm assuming that since I'm powering from buderus unit, the power it delivers isn't enugh, so I have to hook it up via usb and the perform the update, right? The wifi signale is pretty good, so it shouldn't be that: `Signal strength: -55 dB`

If so, then I'm assuming that this would be the steps (correct me if I'm wrong):
 - power off buderus unit (Logamatic 2107 M)
 - change jumper position to usb
 - plug into the computer
 - compile formware and flash

my questions would be:
 - Can I leave the ESP32 module plugged into Logamatic 2107 M during the process?
 - Should the Logamatic 2107 M stay powered off or can I turn it on during the process?
 - Should I adopt also the new safe mode, e.g.:
```
 safe_mode:
  disabled: false
  reboot_timeout: 15min
  num_attempts: 10

```

Btw. thanks for creating this. This module is one of the main reasons I use home assitant at all :)

jensgraef, 2024-06-28:
> 
> btw, did yopu had this error on ota? ` [Errno 104] Connection reset by peer`

Hi Alex,
I observed a similar issue when I tried an OTA update some weeks ago after running the module for half a year without resets.
The device stopped responding. I did a power cycle by turning the whole Buderus of and on again.
Afterwards OTA worked fine.
Maybe there's an issue with doing an OTA update when the device has been running for quite some time.

Does a reset help to get rid of the error?

AlexKusnezov, 2024-06-28:
> Hi Alex, I observed a similar issue when I tried an OTA update some weeks ago after running the module for half a year without resets. The device stopped responding. I did a power cycle by turning [...]

thanks for how long did you had Buderus powered off?
I tried it maybe about 5 times.
tjis is my power graph of the esp32:
[image]

jensgraef, 2024-06-28:
I turned it off for just about a minute and it worked on the second attempt. So it seems you are encountering a different issue.

the78mole, 2024-06-29:
If nothing works, avoid a WiFi connection (shutting down your access point or moving the module away and supply it through USB. After some time, you should be able to access the fallback hotspot and do a FW upgrade through the web UI. Other option is to flash it through serial.

the78mole, 2024-07-12:
I believe, that it is the OTA platfrom settin, introduced in the last esphome update, but I could be wrong:

```
ota:
 - platform: esphome
```

AlexKusnezov, 2024-07-12:
> I believe, that it is the OTA platfrom setting, introduced in the last esphome update, but I could be wrong:
> 
> ```
> ota:
>  - platform: esphome
> ```
> 
> Please check, if the platform line is in your YAML...

yes, it is.

the78mole, 2024-07-13:
It looks like this problem is somewhere hidden in the esphome itself: https://github.com/esphome/esphome
I really have no clue, why this happens.

AlexKusnezov, 2024-07-13:
> It looks like this problem is somewhere hidden in the esphome itself: https://github.com/esphome/esphome I really have no clue, why this happens.

got you, sre you still on lower than 2024.6, too?

the78mole, 2024-07-13:
I'm currently on 2024.6.2, so I integrated the esphome platform statement in my YAML and it works. But did not update to 2024.6.6 yet.

AlexKusnezov, 2024-07-13:
ok, I'm still on 2024.5. I'll give it a try next week

AlexKusnezov, 2024-08-29:
guys, I finally managed to update my untit to 2024.8.1.

What I did was shutting down Home Assistant, putting the Logamatic 2107 M into maintenance mode, so it would heat all the time (this way current goes up) and then try for like 20+ times to upload the firmware. Often it would just cancel on approx 20% upload progress.

AlexKusnezov, 2024-09-12:
2024.8.3 - update worked seamless👍

the78mole, 2024-09-12:
That sounds great. I had this wired issues with FW OTA a looooong time ago (at least one year) and it seems, that is was mostly related to software, more precisely the Arduina-OTA package. I tried many times, but it did not work. Then I just forgot about it and some weeks later, it magically worked.

closed 2024-09-12 by the78mole