the78mole/esphome_components #79: Integration and Firmware Update

tefod-zz, 2024-12-13:
Hi,
First of all, thank you for this fantastic project! I received the board in March 2023, but due to time constraints, I’m only now getting around to integrating it. Setting it up via the Captive Portal in Home Assistant worked like a charm, so thank you for making this so accessible.

> If you selected ESPhome, you will get an (older) ESPhome firmware version for this board. But as you will update the firmware after adjusting the config, you will have the latest firmware aligned with your ESPhome instance. 

I have a few questions about the firmware and have gotten a bit lost navigating the different pages ([1](https://the78mole.de/projects/km271-wifi-howto/#ESPhome),[2](https://github.com/the78mole/km271-wifi),[3](https://github.com/the78mole/esphome_components)):

- Current Firmware: Which is the latest firmware for ESPHome, and where can I find it? Is it located in [this repository]Is it [that](https://github.com/the78mole/esphome_components/tree/gh-pages/km271-for-friends-esp32) or [this](https://github.com/the78mole/km271-wifi/releases/tag/0.0.7)?
- Web Frontend: After connecting to the Wi-Fi, I no longer have access to the web frontend. I’d like to update the firmware via OTA, but since Port 80 is no longer accessible, I’m unsure how to proceed. Can I perform the OTA update through Home Assistant, or is there another method I should follow?

Thanks in advance.
Greetz

the78mole, 2024-12-13:
After setting up the WiFi credentials through the captive portal, the device should show up in home assistant, if ESPhome add-on is installed. From there you can adjust the yaml config of the board and install the new firmware over the air. Everything (generating the code, compiling and downloading it over-the-air to the device) will be handled automatically by ESPhome. You don't need the web frontend or the latest firmware from GitHub. It will be compiled freshly by ESPhome add-on.
Within the YAML you can enable some WebFrontend again, but this is not needed for the module to work...

tefod-zz, 2024-12-13:
Thx for ur quick. The confusion came from the "ESPHome" add-on [now being called](https://youtu.be/Az1YNjE7A2c?t=73) "ESPHome Device Compiler", which must be installed separately. Home Assistant still finds the integration without it, but I couldn’t edit the YAML or update the firmware until the add-on was installed. After that, everything worked perfectly.

Two quick suggestions for the documentation:
- Under ota:, the platform must be specified, or the installation fails.
[image]

- A note or screenshot for locating the KM271 slot would be super helpful.
[image]

Thanks again.
Regards,

the78mole, 2024-12-14:
Hi @tefod-zz,

I know about the platform "problem", but I did not have time to update the docs yet. It's not a single file to edit, it's distributed through different snippets an d configs... But I#ll do that soon. 
On my blogpost, you could find some images about the location :-)

Thanks and Regards,
Daniel