the78mole/esphome_components #80: Potential Mapping Issue with Flow Temperature (Vorlauftemperatur)?

tefod-zz, 2024-12-16:
Hi,
I noticed that the value displayed for"HK1 Vorlauftemperatur" _(HC1 Flow Temperature)_ is the same as for "Kesselvorlauftemperatur" _(Boiler Actual Temperature)_. However, on the heating system's display, both values are shown correctly and are distinct.

[image]

I’m not sure if I might have configured something incorrectly or if there could be a mapping issue in the source code.

the78mole, 2024-12-16:
Hi tefod-zz

Interesting... The parameters are different.
[km271_params.h#L21](https://github.com/the78mole/esphome_components/blob/9d74d117d01fa6e8576c2e30f0d812f270c2a6d8/components/km271_wifi/km271_params.h#L21)
[km271_params.h#L64](https://github.com/the78mole/esphome_components/blob/9d74d117d01fa6e8576c2e30f0d812f270c2a6d8/components/km271_wifi/km271_params.h#L64)

And I can not find a bug in const.py or the params.h...

Could you check your YAML, if the identifier is correct?

Maybe this is a Software Problem of the Buderus control unit???

Regards,
Daniel

tefod-zz, 2024-12-16:
Hey Daniel,
Thanks for your quick reply.
[image]
It’s not related to the YAML.
I also suspect this issue comes from the Buderus CU – it’s at least 13 years old and hasn’t received any updates so far. To be honest, I don’t really know how to update it or if it would even be a good idea. :)

I’ve got a few DS18B20 sensors lying around here. I’ll attach them to the pipes (Vorlauf and Rücklauf) and connect them to an ESP32-C3. 

Thanks & Regards,

Bascht74, 2024-12-28:
If the heating does not have a mixer, it is normal in my opinion that the Brennertemp = HK1-Vorlauftemp flow temp.

@the78mole Can you publish a new version with the fixes / enhancements since 2023 (V1.0.0)? That would be great!!!

dewenni, 2024-12-29:
I can also confirm that in a simple Setup (only HK1):

- HK1_Vorlaufisttemperatur = Kessel_Vorlaufisttemperatur

[image]

- HK1_Vorlaufsolltemperatur != Kessel_Vorlaufsolltemperatur

[image]

tefod-zz, 2024-12-29:
> If the heating does not have a mixer, it is normal in my opinion that the Brennertemp = HK1-Vorlauftemp flow temp.
With a mixer and underfloor heating (FBH), my boiler temperature reachs max 90°, while the flow temperature (Vorlauf) rarely exceeds 40°. The different temperatures are also visible on the control unit display, so I can definitely imagine, as Daniel mentioned, that the Buderus controller has a bug here 🤷‍♂️. 

That's not a problem. I'll attach two external DS18B20 sensors to the heating pipes for the flow and return temperatures and integrate everything into Home Assistant using ESPhome/Tasmota.

the78mole, 2024-12-29:
> @the78mole Can you publish a new version with the fixes / enhancements since 2023 (V1.0.0)? That would be great!!!

No idea, how shall I apply a fix here...

Bascht74, 2024-12-29:
> > @the78mole Can you publish a new version with the fixes / enhancements since 2023 (V1.0.0)? That would be great!!!
> 
> No idea, how shall I apply a fix here...

No, I ment with all the fixes and enhancements since the first release 2023 (V1.0.0), not the one here :-)

jensgraef, 2024-12-30:
@tefod-zz The buderus control unit sends messages when a sensor value changes.
We ignore all messages for sensor values that we don't have a mapping for. So maybe the unit sends the sensor value but we just don't do anything with it...
If you have some time to investigate further, maybe we can find a way to do get the temperature values without additional hardware:

What you'd need to to is to look at the ESPHome logs and see if there's a message with the value you are interested in.
The easiest way to do that is using the ESPHome dashboard: https://esphome.io/guides/getting_started_command_line.html#bonus-esphome-dashboard

It should look like this:
[image]

So if you can find a line with the value you are missing, we can add a mapping for it in this software.

But if you are happy to add a DS18B20, feel free to do that. As the saying goes: The experienced software developer opts for a hardware solution. 😄

tefod-zz, 2024-12-30:
@jensgraef 
thx a lot for the tip and for pointing into the right direction! Obviously sw solutions are often being better—no need to solder when you don’t have to. (Measure twice, cut once, right😉)

I’m much more familiar with Tasmota than ESPHome, but your advice with the logs helped me find the value right away. It seems to be 0x8115, as I simultaneously observed 38°C on the heating control, which corresponds to 0x26 in Hex. Interestingly, this value is labeled for HK2, which isn’t active for me. The mixer values now also make sense—despite having just one mixer, it seems mapped as well to Mixer2.

I’ve adjusted the YAML accordingly:
`sensor:
  - platform: km271_wifi
#    heating_circuit_1_flow_temperature:
    heating_circuit_2_flow_temperature:
      name: "HK1 Vorlaufisttemperatur"
#    heating_circuit_1_mixer_position:
    heating_circuit_2_mixer_position:
      name: "HK1 Mischerstellung"`

It works perfectly now. Thanks again!

[image]

jensgraef, 2024-12-31:
Great!

closed 2024-12-31 by jensgraef