the78mole/esphome_components #89: ESPHome 2025.7.0; Failed config

hergie1976, 2025-07-17:
Nach dem Update von ESPHome auf 2025.7.0 und anschließendem Compilier-Vorgang der Karte habe ich folgende Fehlermeldung bekommen:

`INFO ESPHome 2025.7.0
INFO Reading configuration /config/esphome/buderus-heizung.yaml...
INFO Updating https://github.com/the78mole/esphome_components.git@main
INFO Updating https://github.com/the78mole/esphome_components.git@main
WARNING GPIO2 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
Failed config

number.km271_wifi: [source /data/packages/040b63aa/components/km271_wifi/km271-for-friends.yaml:306]
  platform: km271_wifi
  config_ww_temperature: 
    name: Warmwassersolltemperatur Tag
    mode: slider
  config_frost_switch_temperature: 
    name: Frost Umschalttemperatur
    mode: slider
  config_heating_circuit_1_room_target_temperature_night: 
    name: Heizkreis 1 Raumsolltemperatur Nacht
    mode: slider
  config_heating_circuit_1_room_target_temperature_day: 
    name: Heizkreis 1 Raumsolltemperatur Tag
    mode: slider
  config_heating_circuit_1_holiday_target_temperature: 
    name: Urlaubstemperatur
    mode: slider
  config_heating_circuit_1_flow_temperature_max: 
    name: Heizkreis 1 Heizkreistemperatur Maximal
    mode: slider
  config_heating_circuit_1_design_temperature: 
    name: Heizkreis 1 Auslegungstemperatur
    mode: slider
  config_heating_circuit_1_room_temperature_offset: 
    name: Heizkreis 1 Raumtemperatur Offset
    mode: slider
  config_heating_circuit_1_holiday_days: 
    name: Heizkreis 1 Urlaubstage
    mode: slider
  config_heating_circuit_2_room_target_temperature_night: 
    name: HK2 Raumsolltemperatur Nacht
    mode: slider
  config_heating_circuit_2_room_target_temperature_day: 
    name: HK2 Raumsolltemperatur Tag
    mode: slider
  
  Duplicate number entity with name 'Urlaubstemperatur' found. Each entity must have a unique name within its platform across all devices.
  config_heating_circuit_2_holiday_target_temperature: 
    name: Urlaubstemperatur
    mode: slider
  config_heating_circuit_2_flow_temperature_max: 
    name: HK2 Heizkreistemperatur Maximal
    mode: slider
  config_heating_circuit_2_design_temperature: 
    name: HK2 Auslegungstemperatur
    mode: slider
  config_heating_circuit_2_room_temperature_offset: 
    name: HK2 Raumtemperatur Offset
    mode: slider
  config_heating_circuit_2_holiday_days: 
    name: HK2 Urlaubstage
    mode: slider`

hergie1976, 2025-07-17:
Was können wir denn machen? Oder muss im Code was passieren?

AlexKusnezov, 2025-07-17:
> Was können wir denn machen? Oder muss im Code was passieren?

Ich denke bezgl. den decprecations das:
`legacy adc calibration driver is deprecated, please migrate to use esp_adc/adc_cali.h and esp_adc/adc_cali_scheme.h`

und bzgl. The program size (1950252 bytes) is greater than maximum allowed (1835008 bytes) könnte daran liegen dass ich
bei mir in der yaml bluetooth proxy drin haben. Ich werde es mal ohne probieren, könnte aber duern

AlexKusnezov, 2025-07-17:
@hergie1976 für deinen Fall könnte also diese Änderung reichen:
```
config_heating_circuit_1_holiday_target_temperature:
name: Urlaubstemperatur HK1
```

```
config_heating_circuit_2_holiday_target_temperature:
name: Urlaubstemperatur HK2
```

the78mole, 2025-07-17:
> und bzgl. The program size (1950252 bytes) is greater than maximum allowed (1835008 bytes) könnte daran liegen dass ich bei mir in der yaml bluetooth proxy drin haben. Ich werde es mal ohne probieren, könnte aber duern

Den Bluetooth-Proxy würde ich tatsächlich nur alleine laufen lassen und nicht auf einem ESP integrieren, der "wichtige" andere Funktionen ausführen soll... Wenn ich mich recht erinnere, wird ohnehin empfohlen, als Framework dann auch ESP-IDF zu nehmen... Und das verträgt sich oft nicht gut mit manch anderen Komponenten.

the78mole, 2025-07-17:
> Könntest du dir bei Gelegenheit bitte meine Config-Datei anschauen, und mir mitteilen, welche Optimireungen noch benötigt werden (bl proxy nehm ich taus, das ist klar). Das wäre mega.
> 
> [buderus-heizung yaml.txt](https://github.com/user-attachments/files/21293187/buderus-heizung.yaml.txt)

Sorry, ich hab momentan so viel um die Ohren, das geht wirklich nicht.

hergie1976, 2025-07-17:
Alex, wo finde ich diese Datei?

AlexKusnezov, 2025-07-17:
> Alex, wo finde ich diese Datei?

Das ist die Ausgangskonfiguration:
https://github.com/the78mole/esphome_components/blob/main/components/km271_wifi/km271-for-friends.yaml
Diese habe ich dann für mich angepasst.
Du müsstest bei dir aufs ESPHome Daschboard und dann deine Konfiguration dementsprechend editiren "vorher backup runterladen natürlich":
[image]
dann einfach wo zweimal `"Urlaubstemperatur"` steht entweder bei beiden oder beim letzen ein anderen Namen vergeben, z.b. `"Urlaubstemperatur HK2"`

AlexKusnezov, 2025-07-17:
> > Alex, wo finde ich diese Datei?
> 
> Das ist die Ausgangskonfiguration: https://github.com/the78mole/esphome_components/blob/main/components/km271_wifi/km271-for-friends.yaml Diese habe ich dann für mich angepasst. Du müsstest bei dir aufs ESPHome Daschboard und dann deine Konfiguration dementsprechend editiren "vorher backup runterladen natürlich": <img alt="Image" width="1586" height="259" src="https://private-user-images.githubusercontent.com/33514406/467502072-2591d948-8263-48a7-90c8-15b514619911.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTI3NTAwMTgsIm5iZiI6MTc1Mjc0OTcxOCwicGF0aCI6Ii8zMzUxNDQwNi80Njc1MDIwNzItMjU5MWQ5NDgtODI2My00OGE3LTkwYzgtMTViNTE0NjE5OTExLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA3MTclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwNzE3VDEwNTUxOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTU0N2MyOGVjNTBlNDM5YWZmMmIzNTAyMjBmMWY1NWYyOWVkNmFmMDYwM2VhZTM0ZTVhMzZmNzEyOTUzYTAzMTkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.BbOP-IH8WLF1KIKnuqOvBk6Bg0kC3BvAT4tbTp7Uv6w"> dann einfach wo zweimal `"Urlaubstemperatur"` steht entweder bei beiden oder beim letzen ein anderen Namen vergeben, z.b. `"Urlaubstemperatur HK2"`

@hergie1976 war das die Lösung deines Problems?

hergie1976, 2025-07-17:
Bin noch nicht gazu gekommen... 

Schau ich mich nachher noch an.

hergie1976, 2025-07-17:
OK, Also muss ich den Inhalt der KM271-for-friends.yaml (fast) komplett in die buderus-heizung.txt kopieren und dann entsprechend anpassen? Oder reicht es, die vier Zeilen einzufügen?

AlexKusnezov, 2025-07-17:
> OK, Also muss ich den Inhalt der KM271-for-friends.yaml (fast) komplett in die buderus-heizung.txt kopieren und dann entsprechend anpassen? Oder reicht es, die vier Zeilen einzufügen?

Nichts von Beidem, schau einfach wo bei dir in **deiner** Konfigurationsdatei (im ESPHome Dashboard) zwei Mal `"Urlaubstemperatur"` verkommt und ändere eins davon ab.
Dann speichern und validieren, wenn ein grüner Hacken kommt dann ist es gelungen.

AlexKusnezov, 2025-07-17:
Ok, jetzt verstehe ich die Problematik. Der Rest der Konfigurationsdatei ist bei dir verlinkt.
In dem Fall kopierst du es selbst oder wartest bis @the78mole es bei sich in der Repo angepasst hat

closed 2025-07-17 by hergie1976